NVIDIAGameWorks/PhysX-3.4 #86: PhysX 4 - Cloth simulation

alelordelo, 2018-12-03:
Now that Nvcloth is deprecated, what is the current best solution for cloth? Will 4.0 have a dedicated cloth simulation engine?

sabdulajees, 2018-12-04:
Hi, APEX Cloth is deprecated and is replaced with NvCloth.

Zogrim, 2018-12-04:
I believe alelordelo means that even NvCloth page says "his project is currently **not under active development/maintenance** any more". So of APEX is deprecated and NvCloth in not under development, what is ?

alelordelo, 2018-12-04:
> I believe alelordelo means that even NvCloth page says "his project is currently **not under active development/maintenance** any more". So of APEX is deprecated and NvCloth in not under development, what is ?

That is correct @Zogrim 

APEX - Deprecated
NVCloth - not under active development/maintenance any more.

What is recommended for cloth simulation? Will 4.0 have a dedicated cloth simulation engine?

accounta, 2018-12-05:
Whoa, I didn't realize this "not under maintenance" comment was added to the readme, it is a misunderstanding that I need to clear up.  

We have been working on an update for NvCloth for over a year with the main goal of adding robust support for multi-layer cloth -- basically implementing our "Air Meshes for Robust Collision Handling" research paper.  During this development we realized that while the runtime approach was good, it was really tricky to make a cooking algorithm that would reliably preprocess arbitrary clothing meshes for use by runtime, and that we would need to substantially extend the release timeline of the project to resolve this.  Meanwhile we were hearing from a number of key customers that that they preferred using the cloth simulation in FLEX rather than in NvCloth because they liked the unified solver with soft bodies and liquids that they considered more important.  While we deliberated over this I needed the NvCloth guys to help me expedite progress on some other projects in the near term.

We were all on the same page that it doesn't make sense to have a multilayer cloth feature with unreliable cooking.  After we ship PhysX 4.0 really soon, we can have a discussion where to take NvCloth. I invite anyone interested to email me at amoravanszky@nvidia.com regarding where you would like us to go with it.  Should we move it under BSD like we just did with PhysX?  Would full orthogonality with PhysX features help?  Are you an expert on volumetric mesh processing and can help us fix the cooking issue?  Or just tell me what you'd like to use it for.  We need to gauge interest and balance it against expediting development of other libraries.  We are one team behind all this tech.

a908778387, 2018-12-19:
Is "Air Meshes" major to improve self collision handling? Current version, I found NvCloth may easily found collision penetration on low resolution models.

mtamis-NV, 2018-12-19:
The airmesh approach promises robust recovery after two layers intersect. It is more aimed at multi layer than at self collision. We ran into a lot of problems with the airmesh as we tried to get away with not updating the tetrahedralization at runtime. Maybe it would also work for self collision if the airmesh is updated at runtime, but that would be really expensive.

bgsa, 2019-05-24:
> Whoa, I didn't realize this "not under maintenance" comment was added to the readme, it is a misunderstanding that I need to clear up.
> 
> We have been working on an update for NvCloth for over a year with the main goal of adding robust support for multi-layer cloth -- basically implementing our "Air Meshes for Robust Collision Handling" research paper. During this development we realized that while the runtime approach was good, it was really tricky to make a cooking algorithm that would reliably preprocess arbitrary clothing meshes for use by the runtime, and that we would need to substantially extend the release timeline of the project to resolve this. Meanwhile we were hearing from a number of key customers that that they preferred using the cloth simulation in FLEX rather than in NvCloth because they liked having the unified solver with soft bodies and liquids. While we deliberated over this I needed the NvCloth guys to help me expedite progress on some other projects in the near term.
> 
> We were all on the same page that it doesn't make sense to have a multilayer cloth feature with unreliable cooking. After we ship PhysX 4.0 really soon, we can have a discussion where to take NvCloth. I invite anyone interested to email me at [amoravanszky@nvidia.com](mailto:amoravanszky@nvidia.com) regarding where you would like us to go with it. Should we move it under BSD like we just did with PhysX? Would full orthogonality with PhysX features help? Are you an expert on volumetric mesh processing and can help us fix the cooking issue? Or just tell me what you'd like to use it for. We need to gauge interest and balance it against expediting development of other libraries. We are one team behind all this tech.

@amoravanszky , I have been making a research in "Unified Particle Physics" [Muller2014] and I found out APEX was removed from PhysX. Does it means APEX library will be deprecated or it will be used together with PhysX? Just in case NVIDIA will keep on working on APEX, do you know where can a I found a rich documentation. On NVIDIA pages there are just a simple example and API Reference.

mtamis-NV, 2019-05-27:
APEX is already deprecated.
We don’t have more documentation available aside from https://gameworksdocs.nvidia.com/APEX/1.4/docs/index.html

sschirm, 2019-05-27:
The APEX SDK is already deprecated, and it has been discontinued for quite a while now.

zfergus, 2019-09-19:
Is "Air Meshes for Robust Collision Handling" now available in NvCloth or PhysX?

mtamis-NV, 2019-09-23:
The status of the air mesh feature has not changed, so no it is not available.

RaiaN, 2019-12-04:
> 
> 
> The status of the air mesh feature has not changed, so no it is not available.

Can we still hope cloth tearing feature will be returned to NvCloth one day? :)

mtamis-NV, 2019-12-09:
There are currently no plans to add a cloth tearing feature to NvCloth.