NVIDIAGameWorks/PhysX-3.4 #88: gcc 8.2 compilation errors on linux64

kewur, 2018-12-04:
`building ../../../../PxShared/bin/linux64/libPxPvdSDKCHECKED_x64.so complete!
LowLevelParticles: compiling checked ./../../LowLevelParticles/src/PtDynamics.cpp...
In file included from ./../../LowLevelParticles/src/PtDynamics.cpp:38:
./../../LowLevelParticles/src/PtDynamicHelper.h: In function 'void physx::Pt::updateCellsSubpacket(physx::Pt::SphUpdateType::Enum, physx::PxVec3*, physx::Pt::Particle*, const physx::Pt::ParticleCell*, const PxU32*, physx::PxU32, const physx::Pt::DynamicsParameters&, physx::Pt::DynamicsTempBuffers&)':
./../../LowLevelParticles/src/PtDynamicHelper.h:141:28: error: passing argument 1 to restrict-qualified parameter aliases with argument 2 [-Werror=restrict]
    updateParticleGroupPair(forceBuf, forceBuf, particles, particles,
                            ^~~~~~~~  ~~~~~~~~
./../../LowLevelParticles/src/PtDynamicHelper.h:141:48: error: passing argument 3 to restrict-qualified parameter aliases with argument 4 [-Werror=restrict]
    updateParticleGroupPair(forceBuf, forceBuf, particles, particles,
                                                ^~~~~~~~~  ~~~~~~~~~
./../../LowLevelParticles/src/PtDynamicHelper.h:164:28: error: passing argument 1 to restrict-qualified parameter aliases with argument 2 [-Werror=restrict]
    updateParticleGroupPair(forceBuf, forceBuf, particles, particles, particleIndices + cell.firstParticle,
                            ^~~~~~~~  ~~~~~~~~
./../../LowLevelParticles/src/PtDynamicHelper.h:164:48: error: passing argument 3 to restrict-qualified parameter aliases with argument 4 [-Werror=restrict]
    updateParticleGroupPair(forceBuf, forceBuf, particles, particles, particleIndices + cell.firstParticle,
                                                ^~~~~~~~~  ~~~~~~~~~
./../../LowLevelParticles/src/PtDynamics.cpp: In member function 'void physx::Pt::Dynamics::updateSph(physx::PxBaseTask&)':
./../../LowLevelParticles/src/PtDynamics.cpp:317:45: error: 'void* memset(void*, int, size_t)' clearing an object of type 'class physx::PxVec3' with no trivial copy-assignment; use assignment or value-initialization instead [-Werror=class-memaccess]
    memset(mTempParticleForceBuf, 0, byteSize);
                                             ^
In file included from ./../../LowLevelParticles/include/PtParticle.h:36,
                 from ./../../LowLevelParticles/src/PtDynamics.h:37,
                 from ./../../LowLevelParticles/src/PtDynamics.cpp:30:
./../../../../PxShared/include/foundation/PxVec3.h:49:7: note: 'class physx::PxVec3' declared here
 class PxVec3
       ^~~~~~
./../../LowLevelParticles/src/PtDynamics.cpp: In member function 'void physx::Pt::Dynamics::updatePacket(physx::Pt::SphUpdateType::Enum, physx::PxVec3*, physx::Pt::Particle*, const physx::Pt::ParticleCell&, const physx::Pt::PacketSections&, const physx::Pt::PacketHaloRegions&, physx::Pt::DynamicsTempBuffers&)':
./../../LowLevelParticles/src/PtDynamics.cpp:471:28: error: passing argument 1 to restrict-qualified parameter aliases with argument 2 [-Werror=restrict]
    updateParticleGroupPair(packetForceBuf, packetForceBuf, packetParticles, packetParticles,
                            ^~~~~~~~~~~~~~  ~~~~~~~~~~~~~~
./../../LowLevelParticles/src/PtDynamics.cpp:471:60: error: passing argument 3 to restrict-qualified parameter aliases with argument 4 [-Werror=restrict]
    updateParticleGroupPair(packetForceBuf, packetForceBuf, packetParticles, packetParticles,
                                                            ^~~~~~~~~~~~~~~  ~~~~~~~~~~~~~~~
cc1plus: all warnings being treated as errors
Makefile.LowLevelParticles.mk:212: recipe for target 'build/LowLevelParticles_checked/LowLevelParticles/src/PtDynamics.cpp.o' failed
make: *** [build/LowLevelParticles_checked/LowLevelParticles/src/PtDynamics.cpp.o] Error 1
`

Builds fine most of the way but dies on me.

I'm on Debian 9 with gcc 8.2

kewur, 2018-12-04:
could only get "make debug" to compile successfully, this time using gcc 7.3

AlesBorovicka, 2018-12-04:
Hi, 
 sorry, we dont use latest gcc because of CUDA compilation (we currently compile internally with gcc 4.8, but I think it should compile fine also with gcc 6.1). I will try to check the latest compilers soon and fix it in the next patch update.

kewur, 2018-12-07:
I was able to compile with 6.1 as you suggested, thank you!

closed 2018-12-07 by AlesBorovicka