NVIDIAGameWorks/PhysX-3.4 #91: Novodex API support

a1batross, 2018-12-04:
There is one project, that still uses Novodex API, not PhysX 3. And it works even in 2018.

https://github.com/a1batross/XashXT_original/tree/master/xash/devkit/source_code/phys_shared
https://github.com/a1batross/XashXT_original/blob/master/xash/devkit/source_code/server/novodex.cpp

Will it be opensourced too?

mittorn, 2018-12-04:
many projects still use PhysX 2.x or earlier, which has different API with Nx prefixes

amoravanszky, 2018-12-05:
You call it NovodeX but actually that is NVIDIA PhysX 2.8.3 if you look closely.  It is still over ten years old!  Any reason you aren't just upgrading to latest?  We could spend arbitrary amounts of time open sourcing every old version.

closed 2018-12-05 by amoravanszky