NVIDIAGameWorks/PhysX-3.4 #93: Dynamic scene query broad phase spatial structure corruption.

AndrewMAlvarez, 2018-12-06:
Recently, I encountered an issue where it appeared that the broad phase structure for dynamic scene queries got corrupted. We had a few different jointed actor configurations setup for a couple of our different rigs in game and one of these rigs had several cases where a kinematic RigidDynamic (A) was constrained to another kinematic RigidDynamic (B) which was constrained to the greater system of RigidDynamics representing the character. In this particular case, however, A only had a single PxShape that had the flag eSCENE_QUERY raised on it while B has several PxShapes consisting of shapes with eSCENE_QUERY or eSIMULATION raised. All the actors exploded into infinity when simulated, which is kinda' something I would expect to happen in this situation. The thing that is confusing to me is that the moment a character with this setup was spawned, the entire dynamic scene query broad phase spatial structure seemed to get messed up. The static structure wasn't affected (which makes sense). This was the only difference between this character's setup and all of the other characters' setups that were running around in the level. The end result was that queries issued would sometimes not return a hit on shapes that were occluded by larger shapes. When stepping through the prefiltering code, the smaller shapes wouldn't even appear while the larger, occluding shapes would. The filtering was setup so that the queries wouldn't be report the larger shapes as blocking or touching. After enough shapes got removed from the broad phase, queries would begin to work as expected again, too. Is there a known issue relating to this?

PierreTerdiman, 2018-12-07:
I'm not quite sure I follow, but there's no known issue related to this as far as I can tell. I suppose you wouldn't have an easy repro to share with us?

AndrewMAlvarez, 2018-12-07:
Unfortunately not. I'm not entirely sure of the exact cause of the issue myself. Based on the investigation that I did, it was due to having a few instances of having 2 kinematic PxRigidDynamic actors constrained where one of them had only a shape with a the simulation flag raised attached to it and the other had only a shape with the scene query flag raised attached. The actor with the shape with the simulation flag raised was constrained to a larger system of constraints that represented the ragdoll/hit detection for one of our characters. This was the only difference I could find between the character that was causing this to happen and the characters that weren't. It may have also been related to the calling setKinematicTarget on the actors in the constrained system as well.