NVIDIAGameWorks/PhysX-3.4 #96: Question of MBP algorithm

ka1nong, 2018-12-12:
Since MBP is a multi-SAP implementation, why MBP does not perform as well as SAP when most objects sleep. My scene is too big, there are very few dynamic objects, I want to use MBP to make the performance better.

PierreTerdiman, 2018-12-13:
Different algorithms have different performance characteristics.

A traditional SAP implementation takes no time at all when all objects are sleeping, so nothing will beat SAP in that case.

MBP borrows ideas from multi-SAP but it doesn't use a traditional SAP implementation under the hood. As a result, it has a constant management overhead that we have to run even when all objects sleep. It will be slower than SAP in that case. On the other hand it will be much faster than SAP at the other end of the spectrum, when all objects are moving. In other words, SAP has a better "best case" but a worse "worst case".

Now when "most objects sleep", it's hard to tell which one will be faster. It will depend on the scene configuration, how much objects are moving, etc, etc.

It sounds like you did try MBP already. How did you set it up? How are the regions defined? How many regions did you use? Performance might get better if you tweak this.

But it is also possible that MBP will not be faster if you really have only a few objects moving around.

Note that we have a new broadphase implementation coming in 4.0, that you could also eventually try.

akaStiX, 2018-12-19:
FYI, we had a performance spikes with SAP in some cases when we were spawning new enemies in Vermintide 2. It was quite severe on consoles. But when I tried MBP it was gone

bgsa, 2019-05-29:
@PierreTerdiman when you say? "a new broadphase implementation is coming", do you know what algorithm it is based ?