NVIDIAGameWorks/PhysX-3.4 #101: APEX for arm64-v8a

chege011, 2018-12-29:
Is there any possible to compile APEX for android arm64-v8a correctly? 
I build arm64-v8a with cmake and compile successful. But when I integrate it into UE4, there is something wrong with cloth simulation.

By the way, when I use armeabi-v7a with NEON, the simulation is correct. So, if there is something wrong with arm64-v8a without definding __ARM_NEON__?

I try to add __ARM_NEON__ when compiling arm64-v8a, but I got a compilation wrong.

chege011, 2019-01-02:
I manage to got this thing done.

closed 2019-01-02 by chege011

chege011, 2019-01-07:
I manage to compile apex arm64-v8a without neon. However, it seems that cloth simulation using nvcloth is very slow on Android. 
I found that it uses Scalar4f without NEON and Simd4f with NEON. Is that relevant to the speed of cloth simulation?
If so, how can I successfully compile arm64-v8a with NEON.

mtamis-NV, 2019-01-08:
I'm not sure if you are talking about NvCloth or APEX now.
It is expected that disabling NEON would cause a slow down, as those are the simd instructions used on arm to improve floating point performance.

mtamis-NV, 2019-01-08:
I would suggest using the default UE4 integration of NvCloth. If that still causes problems you should file a bug with Epic Games.

chege011, 2019-01-08:
Default UE4 nvcloth does not support android.I have to upgrade nvcloth and apex to the newest version. What I am trying to say, if there is anyway I can compile Nvcloth and Apex right on arnv8 with neon? Is there any specific macro I need to add? I have already make them right for android on armv7 with NEON and armv8 without NENO, only left for armv8 with NEON.

mtamis-NV, 2019-01-08:
The default config of NvCloth should compile for armv8 with NEON if I remember correctly.
Are you trying to update the existing UE4 integration?
Did you ask Epic Games about running clothing on android?

chege011, 2019-01-08:
You are right, defaut is armv8 with neon. I 
For EPIC, no, I have no idea where to ask. 
Is there any way I can send my research to your guys? It will be easier to ask question. I changed some files in order to compile armv8 with neon for APEX. However, I am not totally sure that I change them in the right.(Not compilation error, but maybe logical error).

mtamis-NV, 2019-01-08:
I'm only responsible for NvCloth.
Apex is deprecated, so I don't think we are taking any changes to that anymore.
Maybe someone else can comment in more detail about APEX. @sschirm ?

sschirm, 2019-01-09:
I second that, we can't really provide support for porting APEX to newer platforms unfortunately. 
What are you using APEX for, if not for clothing?

chege011, 2019-01-09:
I am just trying to integrate APEX in UE4. And I have checked that UE4 use APEX for storing Cloth data(maybe, but I am sure it need APEX in Cloth Simulation). 
I have already compiled armv7 with neon. When I try to compile armv8 with neon, I have to change APEX_1.4\module\clothing\embedded\LowLevelCloth\src\neon\SwCollisionHelpers.h. In line 54 and 75 with __arm64__ macro, function vtbl1q_u8 and vtbl2q_u8 are not found in 
android-ndk-r12b\toolchains\aarch64-linux-android-4.9\prebuilt\windows-x86_64\lib\gcc\aarch64-linux-android\4.9.x\include\arm_neon.h.
So, I comment the macro for armv8 with neon compilation.
Does this change have some bad impact on cloth simulation?

chege011, 2019-01-10:
Let me make my problem clearly. 
First, when I integrated armv7a with NEON(include APEX and NvCloth) in UE4, all things were right. 

Second, when I integrated armv8a with NEON(only NvCloth compiled with NEON, APEX compiled without NEON), I found something wrong. 
1. Cloth without bones and animation simulated and appeared right.
2. Cloth with bones and animation simulated wrong.

Third, when I integrated armv8a without NEON(NvCloth without NEON, with SwSolver.cpp changed and use SwSolverKernel<Scalar4f>(*mCloth, data, allocator, factory)(), APEX compiled without NEON), this time, everything are back to the right place(simulated right and appeared right).

Accroding to those phenomenons, I am totaly confused about why it is not working in Second case. Is there something logically difference between armv7a with NEON and armv8a with NEON?

sunhanpt, 2019-11-14:
Have you solved the problem? I have the same as you..

chege011, 2019-11-14:
Is the armv8a with NEON problem?
I didn't solve that.

By the way, I manage to make armv7a with NEON and armv8a without NEON right.

sunhanpt, 2019-11-14:
oh.. the armv8a with NEON can not simulate correctly. The cloth mesh disappear. 
armv7a with NEON is right.
I will try armv8a without NEON.. 
Thanks.

chege011, 2019-11-14:
By the way, I was using UE4 4.21 at that time.

shlzxjp, 2022-04-30:
@chege011 
Hi，Is your problem finally solved?
I upgraded nvcloth to the latest version, there is no effect of cloth simulation on Android? Only met?

yanfeiwong, 2022-05-23:
> @chege011 Hi，Is your problem finally solved? I upgraded nvcloth to the latest version, there is no effect of cloth simulation on Android? Only met?

I think only deal with nvcloth might not be able to solve the cloth simulation issue... just as above mentioned above... and I have tried all day, no luck to compile apex for arm64... hope there will be a solution ...

komarevtsevdn, 2024-04-29:
Hello, Sorry for up an old post but have you found any solution to compile clothing simulation for android ?

@yanfeiwong  @sschirm @LoneRangerY @chege011

komarevtsevdn, 2024-11-01:
@shlzxjp 

Hello, have you found a solution for that ?