NVIDIAGameWorks/PhysX-3.4 #104: Unable to joint a RigidStatic with a CharacterController

tehmanu, 2019-02-23:
Hi!,

I'm triying to simulate a Railgun for a shooting game, that means that a shot enemy entity ( a physx character controller actor) will be pinned up to a wall (rigid static actor). I'm trying to use an unbreakable FixedJoint, that should be pretty simple:

PxFixedJoint* joint = PxFixedJointCreate(*physics,
			enemyActor, enemyTrans,
			wallActor, wallTrans);

where enemyTrans is the hit point on the enemy in world coordinates and wallTrans is the projection of that point on the wall (found by a second ray). The point is that the joint does simply not work, since after the lock, the enemy is able to move.

Any ideas for this? Thank you!

kstorey-nvidia, 2019-02-25:
Many character controllers (including the CCT implementation that comes with PhysX) use kinematic actors to represent the character. This means that the character's rigid body has infinite mass and goes wherever it is commanded, pushing everything out of the way. The character update code performs sweeps to ensure that it doesn't go through the static environment, other characters or dynamic objects.

As the character has infinite mass, it is not possible to constrain it to the static environment with a joint. You can use a joint to attach things to the character but, as the character has infinite mass, its motion will not be affected by the joint.

tehmanu, 2019-02-25:
Oh, i thought it right then. So is it possible to dynamically change a character controller into a dynamic object on the fly? ( I guess that is how the ragdolls work?)

thank you very much for the answer!