NVIDIAGameWorks/PhysX-3.4 #105: Collision Detection between 2 Kinematic Objects

hangoon18, 2019-03-19:
Hello, My name is Kim.

The reason why I question to you because I have a problem with my project.

My Project is now using PhysX 3.4.

I have 2 objects which one is kinematic object(shape is hand) and another is dynamic object(shape is box).

when I contact dynamic object with my kinematic object, then I change dynamic object's collision type into kinematic.

Here is the problem. I can't get Collision Detection with 2 kinematic objects.

I also use filtershader and contactModifiyCallback in my SceneDesc.
(I set up PxSceneFlag::eENABLE_KINEMATIC_PAIRS flag, but I got a new error "<893> : warnig : Filtering : Resolving contacts between two kinematic objects is invalid. Contacts will not get resolved.")

I read on the internet, physX doesn't support Collision Detection between 2 kinematic objects.. is that true?

or is there any solution to fix this problem?

I hope I can get a solution..

thank you for reading

nvidia-bgaldrikian, 2019-03-19:
Hi Kim,

Setting the eENABLE_KINEMATIC_PAIRS flag causes pair data to be sent to your filter shader.  But if you return from the shader with eSOLVE_CONTACT in your pairFlags, you'll get that warning.  Something like

`	// eSOLVE_CONTACT is invalid with kinematic pairs
	if (PxFilterObjectIsKinematic(attributes0) && PxFilterObjectIsKinematic(attributes1))
	{
		pairFlags &= ~PxPairFlag::eSOLVE_CONTACT;
	}
`

will ensure that you're not requesting contact resolution from kinematic-kinematic pairs.  You'll still want your pair flags to contain notify flags (such as eTRIGGER_DEFAULT), so that your onContact callback will receive the pairs.

Bryan Galdrikian

hangoon18, 2019-03-20:
Hello Bryan.

I read your solution, but I can't get the point.. sorry.

some informations that I check with your solution, first I'm using filter shader with eSOLVE_CONTACT.

And I cant't understand below lines

> Something like
if (PxFilterObjectIsKinematic(attributes0) && PxFilterObjectIsKinematic(attributes1)) { pairFlags &= ~PxPairFlag::eSOLVE_CONTACT; }
will ensure that you're not requesting contact resolution from kinematic-kinematic pairs.You'll still want your pair flags to contain notify flags (such as eTRIGGER_DEFAULT), so that your onContact callback will receive the pairs.

I tried a lot of  things, but I can't exactly what's your mean.
I show my filtershader code. 
Could you find some mistakes to use Kinematic Objects Collision Detection?

`	PX_UNUSED(attributes0);
	PX_UNUSED(attributes1);
	PX_UNUSED(filterData0);
	PX_UNUSED(filterData1);
	PX_UNUSED(constantBlockSize);
	PX_UNUSED(constantBlock);

	if (physx::PxFilterObjectIsKinematic(attributes0) || physx::PxFilterObjectIsKinematic(attributes1))
	{
		pairFlags |= physx::PxPairFlag::eTRIGGER_DEFAULT;
	}

	if (physx::PxFilterObjectIsTrigger(attributes0) || physx::PxFilterObjectIsTrigger(attributes1))
	{
		pairFlags |= physx::PxPairFlag::eSOLVE_CONTACT;
		pairFlags |= physx::PxPairFlag::eNOTIFY_CONTACT_POINTS;
		pairFlags |= physx::PxPairFlag::eNOTIFY_TOUCH_FOUND;
		pairFlags |= physx::PxPairFlag::eTRIGGER_DEFAULT;
	}

	//// generate contacts for all that were not filtered above
	pairFlags |= PxPairFlag::eDETECT_DISCRETE_CONTACT;
	
	////pairFlags |= PxPairFlag::eDETECT_CCD_CONTACT;
	//
	//// trigger the contact callback for pairs (A,B) where 
	//// the filtermask of A contains the ID of B and vice versa.
	if ((filterData0.word0 & filterData1.word1) && (filterData1.word0 & filterData0.word1)){
		pairFlags |= PxPairFlag::eSOLVE_CONTACT;
		pairFlags |= PxPairFlag::eNOTIFY_TOUCH_FOUND;
		pairFlags |= PxPairFlag::eNOTIFY_CONTACT_POINTS;
		pairFlags |= PxPairFlag::eMODIFY_CONTACTS;
	}

	return PxFilterFlags();
`
and sceneDesc flag setting code is below.

`sceneDesc.flags |= PxSceneFlag::eENABLE_PCM | eENABLE_KINEMATIC_PAIRS;`

hopefully, If you give me some help, I can fix this problem. 
Thank you for your help again.

nvidia-bgaldrikian, 2019-03-20:
Hi Kim,

If you put the lines

	if (PxFilterObjectIsKinematic(attributes0) && PxFilterObjectIsKinematic(attributes1))
	{
		pairFlags &= ~PxPairFlag::eSOLVE_CONTACT;
	}

just before the return statement in your filter shader, that should do it.

Bryan

hangoon18, 2019-03-27:
Hello Bryan,

I thank you for helping me.

I put your code in my filtershade code. 

then, the function 
`void ContactReportCallback::onContactModify(PxContactModifyPair* const pairs, PxU32 count)`

give me a break trigger.

in the function,
`for (PxU32 i = 0; i < count; i++)
		{
			const PxRigidActor** actors = pairs[i].actor;

			const PxActor* actorA = actors[0];
			const PxActor* actorB = actors[1];
			
			const PxU32 nbContacts = pairs[i].contacts.size();
			for (int j = 0; j < nbContacts; j++){
				PxContactSet &pt = pairs[i].contacts;
				Eigen::Vector3f value;
				value[0] = pt.getPoint(i).x, value[1] = pt.getPoint(i).y, value[2] = pt.getPoint(i).z;`

the last line 

> value[0] = pt.getPoint(i).x, value[1] = pt.getPoint(i).y, value[2] = pt.getPoint(i).z;

triggers error, and the programm has shut down.

but, when I erase 

> eENABLE_KINEMATIC_PAIRS
so, `sceneDesc.flags |= PxSceneFlag::eENABLE_PCM;`, it works.

could you tell me what is the problem?

thank you again for your help!

kstorey-nvidia, 2019-03-27:
Hi Kim

There isn't a lot to go on from your description.

Do you have any information explaining what errors are triggered: is it an assert in PhysX or a failure somewhere else, e.g. validating the values passed to the vector? What values are you getting from those contacts?

Also, do you specifically need to modify the contact points? That callback relates to modification (pre-solver) rather than event notification (after solver, during fetchResults). 

Those modification callbacks are executed in-line with the contact generation so they can come from multiple threads simultaneously, whereas the contact notification (onContact) callbacks in PxSimulationEventCallback are guaranteed to be executed on the thread that calls fetchResults.