NVIDIAGameWorks/PhysX-3.4 #107: Back face sweep

Mashewnutz, 2019-05-02:
Hi there,

Are intersections on backface sweeps supported? I'm trying to perform a scene query sweep using sphere geometry inside a box going outwards. My exact setup is:

Box (centered at 7,7,7, dimensions 1,1,1)
Sweep (sphere radius 0.2, pos 7,7,7, direction 1,0,0)

I get a normal back of -1,0,0, but the position is 0,0,0, and the distance is also 0. The flags on the PxLocationHit object are also missing the ePosition flag. 

Is there any way to get a position reported without making the mesh double sided? I tried using the eMESH_BOTH_SIDES flag on the query but it doesn't seem to help.

Thank you!

AlesBorovicka, 2019-05-04:
Hi,
  Do I understand it correctly that you would like to get a sweep position when you do a sphere sweep inside a box collision shape? In this case PhysX does return distance 0.0f, this is because initial overlap happen and in this particular case, each time a distance 0.0f is returned and position is not computed. Note that eMESH_BOTH_SIDES should not help you in this case, as box is not a mesh primitive.

In order to get the position and distance, you need to set this flag: PxHitFlag::eMTD

Regards,
Ales

Mashewnutz, 2019-05-08:
Thanks for your response Ales.

The initialOverlap function on the hit result followed by a further query using the MTD flag should give us what we need.

Thank you!
Matt

closed 2019-05-08 by Mashewnutz