NVIDIAGameWorks/PhysX-3.4 #112: Where is PhysX3.4.0

zhengchuanzhe, 2019-07-23:
I want to use vs2017 to build  PhysX3.4.0, but I can't find it

AlesBorovicka, 2019-07-23:
You mean the solution files for VS2017? They are here:
https://github.com/NVIDIAGameWorks/PhysX-3.4/blob/master/PhysX_3.4/Source/compiler/vc15win64/PhysX.sln

zhengchuanzhe, 2019-07-23:
> You mean the solution files for VS2017? They are here:
> https://github.com/NVIDIAGameWorks/PhysX-3.4/blob/master/PhysX_3.4/Source/compiler/vc15win64/PhysX.sln

This is 3.4.2,But I want 3.4.0

AlesBorovicka, 2019-07-23:
It was not supported during 3.4.0 release because of a compiler bug in VC15, the support was added later. Feel free to upgrade the VC14 projects to VC15.

zhengchuanzhe, 2019-07-23:
> It was not supported during 3.4.0 release because of a compiler bug in VC15, the support was added later. Feel free to upgrade the VC14 projects to VC15.

I build v3.4.0-rc-2  using vs2017,but the rebuilding failed.

zhengchuanzhe, 2019-07-23:
I want 3.4.0，where can I download it?

sschirm, 2019-07-24:
3.4.0 is v3.4.0-rc-2 as far as I can tell. The tag is misleading as it was not updated. But the correct change is "PhysX 3.4, APEX 1.4 final release @21757769" 7840347c32755e94a807f978f1cbb558717ac8cf

closed 2019-07-24 by sschirm