NVIDIAGameWorks/PhysX-3.4 #118: Getting joint motor applied torque

danielkrakowiak, 2019-10-12:
Hi,
is there a way to get torque applied by a motor in a joint?

I also experience a problem where motor set to zero velocity never gets to stop the wheel when that wheel is propelled by friction from the ground. Could I fix that somehow? Is that because the motor only applies the torque needed to stop the rotating mass of given inertia? I hoped that motors have more info on the post-friction velocity and can be more precise then applying and calculating the torque myself. Could you help me understand?

kstorey-nvidia, 2019-10-12:
Are you using articulations or regular d6 joints? For absolute zero velocity (e.g. car handbrake) you might get better results switching to a positional drive that holds at current angle. This should correct any drift.

The applied motor forces aren't currently exposed via the API, but they are recorded internally.

danielkrakowiak, 2019-10-12:
I'm using a D6 joint in Unreal. I guess that positional drive would rotate the wheel backwards a bit after failing to achieve zero velocity quickly. I need to apply the brake smoothly from zero braking torque to maximal braking torque. I don't want to explicitly stop the wheel but rather be able to apply torque high enough to stop it and keep it stopped.

kstorey-nvidia, 2019-10-12:
You could use a velocity drive at higher speed, transitioning to a positional drive when velocity approaches zero to give you a handbrake-like effect. You could put a force limit on the positional drive and detect when velocity exceeds a certain amount to transition to velocity drive again.

danielkrakowiak, 2019-10-12:
I am also wondering about the difference between using positional drive (which seems to require all axes to be free?) and using a normal joint spring or damper for this case of just wanting to slow down or stop the rotation. Could you advise on that?

kstorey-nvidia, 2019-10-12:
There's fundamentally little difference between the two. The positional drive is the same spring-damper. If you don't set a target relative transform, that relative transform is identity and the spring term will drive towards making the two joint frames coincide along/around and driven axes. If you provide a target pose, this is the offset between the joint frames it drives towards. The same effect could be achieved by rotating frames but this can be tricky if you have multiple rotational dofs and joint limits because the joint frames impact those as well.

You don't need all axes to be free/limits - only the ones you want the drive to influence

danielkrakowiak, 2019-10-13:
Thank you for your help and all the suggestions and clarifications! I will try them all out.

closed 2019-10-13 by danielkrakowiak