NVIDIAGameWorks/PhysX-3.4 #125: PxArticulation::applyImpulse - documentation contradictory

philipguin, 2020-02-29:
So looking in PxArticulation at the documentation for methods `applyImpulse` and `computeImpulse`, they're described as affecting the "entire" articulation, while also accepting a `link` parameter described as "the link to which to apply the impulse." What do I expect here? Why does it accept a pointer to `link` instead of a reference? Can I leave it `NULL` to affect the entire articulation as desired (e.g. in the case of gravity?) Or can I only affect one link per call?

kstorey-nvidia, 2020-03-01:
You must provide a link that the impulse is applied to. The impulse applied to that individual link will be propagated through the entire articulation. This can cause a change in velocity of some or all of the links in the articulation, depending on the degrees of freedom of the articulation and the direction of the impulse.

philipguin, 2020-03-01:
I see. So is this is a natural consequence of joints acting as "constraints"? I would normally think of forces as propagating over time via simulation, but I suppose a constraint would mean that the change is propagated instantaneously, by definition. For something like gravity, then, I suppose I would just call "applyImpulse" for each individual link? Sounds like order may be an issue.

kstorey-nvidia, 2020-03-02:
If you apply forces on all links, the resulting velocities should be the same regardless of the order in which you apply them