NVIDIAGameWorks/PhysX-3.4 #133: Considering Optimize Raycast Using Grid Based AABBTree.

gongyiling, 2021-04-28:
as far as i know, PhysX's Raycast using a AABBTree to accelerate Raycast process. this works fine for small or medium scene, but for large scene without streaming(as with the case in Dedicated Server in UE4), there are hundreds of thousands objects which forms an large deep AABBTree, that cause the Raycast performance degenerates.
so is it's possible to split the AABBTree using a 2D grids to optimize performance?

any suggestion is appreciated!

PierreTerdiman, 2021-06-14:
In PhysX 3.x the only way to do that would be using multiple PxScene objects, and manage the 2D grid yourself on top of that. Depending on which PhysX version you are using you might be able to use PxSpatialIndex for this task.

But if your large stream does not stream performance should remain fine, unless your objects all overlap in one place of the map maybe.

"Hundreds of thousands objects" should be fine, unless maybe you have a lot of large objects covering the whole world. In any case managing the 2D grid would add some overhead into the system, and there is no evidence so far that this would provide performance gains overall, especially if you are not streaming.

PierreTerdiman, 2021-06-14:
Actually I forgot one option, you could perhaps look into PxPruningStructure:

https://github.com/NVIDIAGameWorks/PhysX-3.4/blob/master/PhysX_3.4/Include/PxPruningStructure.h

If you have a large static world you might be able to assign these static objects to grid cells, then precompute a pruning structure for each cell, and that could reduce the load on the scene query structure in your case.

gongyiling, 2021-06-19:
> Actually I forgot one option, you could perhaps look into PxPruningStructure:
>
> https://github.com/NVIDIAGameWorks/PhysX-3.4/blob/master/PhysX_3.4/Include/PxPruningStructure.h
>
> If you have a large [...]

this sounds great!

closed 2021-06-19 by gongyiling