NVIDIAGameWorks/PhysX-3.4 #134: solveContact4_StaticBlock function crash

ka1nong, 2021-05-06:
[image]
dynamic actor set  all  material->setFlag(PxMaterialFlag::eDISABLE_FRICTION, true);
the crash is in the place shown in the picture

kstorey-nvidia, 2021-05-06:
Thanks for reporting. Will try to reproduce. Does this also reproduce if you set all friction coefficients to zero but do not raise the eDISABLE_FRICTION flag?

Also, is there scope for upgrading to PhysX 4 (where it is possible this issue was already resolved) or are you stuck on PhysX 3.4?

ka1nong, 2021-05-06:
Continue to use PhysX3.4 version, not 100% reproduced, but the probability is very high, about one-half
hdr->type is zero

kstorey-nvidia, 2021-05-06:
I suspect we are allocating memory for friction constraints in the range currPtr to last, then later setting the friction count to 0 so that block of memory doesn't get written to/read from. The while loop goes round too many times and it is now reading the friction constraint block memory as if it was the next contact header.

It shouldn't be too difficult to fix if that is the case. Ideally, I would still recommend an upgrade to PhysX 4.1 if you can because that includes about 4 years of improvements and bug fixes that are not in PhysX 3.4.

ka1nong, 2021-05-07:
If you have trouble to reproduce, please let me know and confirm that there is a problem with the API and not the wrong way of using it. thank you