NVIDIAGameWorks/PhysX-3.4 #140: Negatively scaled PxTriangleMesh collisions have penetration problems

sunwtGitHub, 2021-12-23:
I now have a step-shaped PxTriangleMesh and set its MeshScale value to PxVec3(-1, 1, 1). This results in a TriangleMesh that is a mirror image of the X-axis. Then roll a small ball down the top of the step. Each time the ball passes the edge of the step, it falls into the TriangleMesh and then pops out of the TriangleMesh. It looks like the edge of the steps didn't collide.

PierreTerdiman, 2022-01-03:
Yes this bug has been fixed in PhysX 5. I am not sure how easy it would be to back-port to 4 or if there are plans to do so. Will ask around.

PierreTerdiman, 2022-01-03:
Hmmm but you're using PhysX 3.4 anyway right? There's certainly no plan to port that fix back to PhysX3 unfortunately.

sunwtGitHub, 2022-01-11:
Thanks for getting back to me. I did the same test with UE4 version 2.7 engine, but there was no similar BUG in UE4. The version of Physx3.4 engine they used seems to be released in 2017, while the version I used was released in 2018. By the way, when will we be able to use PhysX 5 in our own engines?